nisargjhaveri/WirelessAndroidAutoDongle #266: SSH is not available as per documentation

markleigh, 2025-04-24:
When connected via wi-fi, and having enabled ssh in the config file, SSH should be available at 10.0.0.1 with username root and password password. I suspect this is an error with the documentation, rather than a bug. SSH at 10.0.0.1 just times out. The device is visible on the network at 10.0.0.1, and responds to pings, but does not respond to SSH requests. Presumably SSH is running on a non-standard port, but I've been unable to discern which Port it is on. If anyone knows, could it please be added to the documentation so we can get log files?

markleigh, 2025-04-24:
As noted, SSH is enabled in the config file. Is there somewhere else at SSH has to be enabled in order for it to work? If not I guess there's a bug, and I can't provide logs because I can't SSH.

Ioniq3, 2025-04-24:
I access with root@10.0.0.1 without specify any port
The issusñe is to uncomment in the aawgd.conf the SSH key and the wifi password to fix the same during the debug process

manio, 2025-04-24:
@markleigh 
Maybe you are trying to connect while the androidauto is running? You need to disconnect android auto, disable bluetooth for short while and then make explicit connection to the WiFi dongle...

markleigh, 2025-04-24:
> [@markleigh](https://github.com/markleigh) Maybe you are trying to connect while the androidauto is running? You need to disconnect android auto, disable bluetooth for short while and then make explicit connection to the WiFi dongle...

Yes, Bluetooth disabled, wifi connected to the pi (it remains connected after disabling Bluetooth), and still no SSH.

manio, 2025-04-24:
Was it working for you before at some point? Maybe this is some more general problem of ssh client, firewall or something? I didn't have any problems connecting...

manio, 2025-04-24:
As the last resort you can try to connect your Pi via HDMI to some display, attach a keyboard and investigate via console what is going on... if dropbear is running, etc...

markleigh, 2025-04-24:
> Was it working for you before at some point? Maybe this is some more general problem of ssh client, firewall or something? I didn't have any problems connecting...

SSH has never worked. Android auto *sometimes* works temporarily. Is there any way to persist logs via conf file so that I can just pull logs off the SD card?

Ioniq3, 2025-04-24:
Remove /etc/default/dropbear
By this you don't need change the aawgd.conf

jebbgrenham, 2025-04-25:
Thanks @manio; I tried your rust version but it doesn't even broadcast a wifi access point, so I can't SSH into that either. Also (off topic, I know) but your image will only connect to the bluetooth about one in 10 times. I'll try yours again without modifying the conf file to see if it even works as an android auto device, and then try Nisarg's again.

jebbgrenham, 2025-04-26:
Getting back on topic... I tried a new flash; adidtionally un-commenting the wifi password line and setting "1234" as password (instead of just relying on the BT authentication to connect the wifi) and now the unit doesn't even broadcast a wifi AP (same occurred with Manio's version).

jebbgrenham, 2025-04-26:
> Delete /etc/default/dropbear By this you don't need change the aawgd.conf

This will disable SSH I assume, but you're saying this will persist logs?

jebbgrenham, 2025-04-26:
Marking this closed. Not because it's truly fixed (there's clearly numerous bugs), but with huge frustration I was able to get a copy of the logs.

After 5 or 6 SD flashes, and NOT modifying the conf file WiFi password (but un-commenting it), definitely un-commenting the SSH line in conf file, waiting for it to randomly disconnect, then disabling Bluetooth, then manually connecting WiFi using the default password, I was able to make an SSH connection before the WiFi drops out (about 8 seconds later, every time). After many practice runs of quickly connecting the WiFi, switching to SSD client, quickly connecting, typing out "cat /var/log/messages" (there's no bash history apparently...) and then exporting the SSH session to a file all within 8 seconds before the WiFi disconnects, I FINALLY managed to get the log off.

The quick fix here is persisting the logs and using log rotation to not fill the SD card. Someone will be able to fix this easily.

manio, 2025-04-26:
> Thanks @manio; I tried your rust version but it doesn't even broadcast a wifi access point, so I can't even start android auto (or SSH into it either). Also (off topic, I know) but your image will only connect to the bluetooth about one in 10 times. I'll try yours again without modifying the conf file to see if it even works as an android auto device, and then try Nisarg's again.

It is really weird. I have an AP every time the RPi starts and I also don't remind this kind of problems users reported me...
Anyway - as always I cannot tell you more without the logs :(
Maybe this is related with your connectivity problems you reported in #267... no idea...
Also IIRC your Sony HU needs `legacy` flags for working properly with my project, Nisarg's has this enabled by default.

manio, 2025-04-26:
> I was able to make an SSH connection before the WiFi drops out (about 8 seconds later, every time). After many practice runs of quickly connecting the WiFi, switching to SSH client, quickly connecting, typing out "cat /var/log/messages" (there's no bash history apparently...) and then exporting the SSH session to a file all within 8 seconds before the WiFi disconnects, I FINALLY managed to get the log off.

A small tip here: This is a known issue also for me. I think the problem here is when you disable AA in your phone, the Raspberry is still trying to connect to your phone via Bluetooth, but the problem here is that BT and WiFi is interfering (I have a plenty of unanswered pings during that time - because bluetooth connecting seems really "heavy" operation for the bundled BT/WiFi communication chip, especially when target BT device is unavailable). So to avoid this: after you are able to connect, the first thing I am doing is stopping `aawgd`/`aa-proxy-rs` - otherwise it may lead to disconnecting SSH session. I have the `sh` history saved in /root/ so it is much easier to type CTRL-R and command fragment. Depending on ssh client on your phone you can also set initial command set which it is executing upon connect... it's also making things easier.