nisargjhaveri/WirelessAndroidAutoDongle #271: support for other boards?

f4mrfaux, 2025-05-12:
Ihav ebeen attempting to build this for the OrangePi zero2w, a RBp clone board using the breadwinner618. I have been unable to get a working build so far. still trying. hoping the author might be interested in this as well, as the board has better specs, but is slightly cheaper.

manio, 2025-05-13:
Update:
I can see you've ["forked"](https://github.com/f4mrfaux/WirelessAndroidAutoDongle) this and give a ClaudeAI to do some work on this. How it's working?

And regarding this:
@nisargjhaveri this may be worth backporting: https://github.com/f4mrfaux/WirelessAndroidAutoDongle/commit/489ba54ace04353fb2383a00c92ef3676375c3b3

f4mrfaux, 2025-05-13:
It has been rather painful.

I originally attempted to port it by hand a few months ago but wasn't able
to get it to work.

I recently forked pointed claude code at the  debate for both the orange pi
github and yours and tried to see if I could get it to vibe code it with
supervision.

I've got a build that /should/ work but I haven't been able to get a UART
to see what's going on.

I might pivot and get Armbian or dietpi, then try to use your project,
rather than trying to cook it in.

Thank you for your patience, I  by no means a developer, I am using these
homelab projects to learn and make cool and hopefully useful stuff.

 Thanks again

nisargjhaveri, 2025-06-02:
> @nisargjhaveri this may be worth backporting: https://github.com/f4mrfaux/WirelessAndroidAutoDongle/commit/489ba54ace04353fb2383a00c92ef3676375c3b3

Commited as 524dbe68d4f87571b71edb6bb9b12c3dace93134. Thanks @f4mrfaux.

About, Orange Pi boards, it looks interesting. And as long as the boards are supported in buildroot, we should be able to make this work. But as you already know, it can be quite a work ironing out all the quirks.

Hardzen, 2025-07-11:
Has anyone tried the Radxa ZERO 3W RK3566?

https://docs.radxa.com/en/zero/zero3?model=zero-3w#features

manio, 2026-10-17:
FWIW/fyi only: `Orange Pi Zero 2W` support has been added recently to [aa-proxy](https://github.com/aa-proxy/aa-proxy-rs)