nisargjhaveri/WirelessAndroidAutoDongle #272: Optional persistent logs

Jolt151, 2025-05-13:
Hi,

I'm still debugging long connection times with my RPI4 of 1.5 minutes, usually with 1 or two connection failures until a success. I find the process for pulling logs to be very annoying due to the ephemeral nature of them and needing to connect to the Pi's network after it's powered up and connecting to my car just to pull the logs. It would be nice to be enable persistent logs somewhere (along with making the sdcard r/w), even as a build flag so that this can only happen at compile time.

Scg19, 2025-05-19:
`#!/bin/sh
#
# Start aawgd daemon
#

DAEMON="aawgd"
PIDFILE="/var/run/$DAEMON.pid"
LOGFILE="/var/log/$DAEMON.log"

RETVAL=0

case "$1" in
	start)
		if [ -z "$AAWG_CONF_SOURCED" ]; then
			echo "Source /etc/aawgd.conf"
			source /etc/aawgd.conf
		fi

		printf "Starting $DAEMON: "
		start-stop-daemon -S -b -q -m -p "$PIDFILE" -x "/usr/bin/$DAEMON" -- -O "$LOGFILE"
		RETVAL=$?
		[ $RETVAL = 0 ] && echo "OK" || echo "FAIL"
		;;
	stop)
		printf "Stopping $DAEMON: "
		start-stop-daemon -K -q -p "$PIDFILE"
		RETVAL=$?
		[ $RETVAL = 0 ] && echo "OK" || echo "FAIL"
		;;
	restart)
		$0 stop
		$0 start
		;;
	*)
		echo "Usage: $0 {start|stop|restart}"
		;;
esac

exit $RETVAL
`

nisargjhaveri, 2025-06-02:
Yeah, this is a common issue.

We can have a build flag so custom builds can make this easier. We can also try to have a runtime flag in the config file that either symlinks or changes the logs location. One of these options should work, whichever seems the cleanest to have. Though this will come with higher risk of sd card corruption, we can mention that with the config to use sparingly.