nisargjhaveri/WirelessAndroidAutoDongle #273: Help to build dongle (to android auto) Streaming box carpc in Chevrolet Onix with Only AppleCarplay

neuberfran, 2025-05-22:
Hi,

Due to covid 19 my Chevrolet Onix came only with Apple CarPlay and no native Android Auto

I bought CarPlay adapter but it doesn’t connect to GPS (despite built-in GPS in the box)

I have good knowledge of Android and embedded systems and rpi4 and I thought I could make this dongle myself and use a usb cable, connect the rpi4 to my car (I have a galaxy s24 cell phone) and have two essential apps on the screen (which is already built-in) of my Onix 2023: Maps and Waze (with working gps, of course)

I come to this forum to ask for the necessary steps to achieve this in the best way, from software (Operating System, etc.) to hardware, as perhaps the rpi4 is not the best option.

manio, 2025-05-22:
If your Chevrolet only supports CarPlay, this project would not work because it is for AndroidAuto.

neuberfran, 2025-05-22:
@manio I understand your point of view, but I have something to tell you that streaming box type products get around this and allow me to have Android 13 on my car screen. However, I don't have GPS.

Despite this, I need to know things that the streaming box has that I can replicate on the rpi4/5. Like what operating system and what software that uses to make contact with (and how) android auto despite being appleCarPlay.

Tks, and I wait for more information

manio, 2025-05-22:
Sorry, but I still don't understand your plan. Am I correct that you want to wrap AndroidAuto in a CarPlay protocol, like some dongles does?

neuberfran, 2025-05-22:
@manio Yes. https://h5.clewm.net/?url=qr71.cn%2FobX8r5%2Fq0hCSD1 . Pls help-me in this

manio, 2025-05-22:
I am afraid that it is out of scope of this AndroidAuto specific project, moreover for doing anything with CarPlay, you need to have MFi chip, which is mandatory, as this chip is performing verification if the CarPlay headunit is Apple certified. As you can see it rather not doable on simple DIY Raspberry Pi board.

... but of course @nisargjhaveri surely have a last word on this :)

closed 2025-06-25 by neuberfran