nisargjhaveri/WirelessAndroidAutoDongle #281: Multi-device selection by button or switch or priority list

MisterAnderson91, 2025-06-10:
Hey, I'm wondering if it's possible to implement multi-device support to allow any of the a list of phones to connect, but also prioritize a specific one. I sometimes drive my partner's car and I currently just use the radio, as pairing up my own phone to Android Auto seems to cause hers to stop auto-connecting. I believe WirelessAA Two implemented a button to switch device priorities.

Currently I'm not sure of how the connection priority is decided when multiple devices with previous connections are present. I presume that it's one the following:

- Car headunit connects via bluetooth to its preferred device, and that device gets the wireless AA device
- Car headunit connects via bluetooth but the wireless AA device overrides that and switches to its own preferred device
- the wireless AA device handles everything

I'm thinking if it's the second or third, we could implement a selector via the GPIO pins (momentary switch or physical switch) that could be read and will prioritize the selected device. Could be done with a bootloader and two separate copies of the OS selected at startup, or by shifting config files around, or even just changing the wifi-AP name and pairing devices to different names.

Thoughts?

manio, 2025-06-10:
Similar advanced work for scripting and GPIO pin support work was done by @Ioniq3. Join us on [discord](https://discord.gg/c7JKdwHyZu) if you like, I am pretty sure he would give you advice or maybe even provide a full script.

nisargjhaveri, 2025-06-10:
In the default phone first connection strategy, the dongle handles everything. It tries to connect to first avialable device via Bluetooth (in undefined order) and whichever device connects first gets wireless aa connected.

The dongle mode however should have a better control over this. In the dongle mode, the dongle just tries to connect to all previously connected bluetooth device. Whichever device is connected to the car headunit gets the wireless aa in this case. This should give you much better control for which device should connect.

If that does not work, we can implement configurations for priority, or switching on external signal like a button via gpio. Might be some woke to make the switching stable and fast enough though.

MisterAnderson91, 2025-06-10:
I'm not worried about actually switching device during use, mostly only at first start. Switching while in use would be great though. But a priority list or selector is probably most important because there is no simple way to prefer her phone over mine, meaning when we ride together in the car it's going to be random which gets connection. As a workaround, I can just unpair my phone after use.

Luen, 2025-06-11:
An alternative workaround would be to turn your Bluetooth off, instead of unpairing your phone.

manio, 2025-06-11:
The script I was mentioning earlier is covering multiple use cases: wireless, wired only version (without bluetooth), disabling AA and enabling mass-storage (for playing music directly from rpi SD-card media partition), forcing restart, etc...
@Ioniq3 is a smart guy ;)
It mainly working by restarting daemon but with different parameter set.

But if you only need to connect to specified device it should be even easier using my [aa-proxy-rs](https://github.com/manio/aa-proxy-rs) because I have this [parameterized](https://github.com/manio/aa-proxy-rs?tab=readme-ov-file#usage):
`-c OP:TI:ON:AL:MA:CA:DD:RE:SS`
So you should be able to connect to your phone or your wife's phone depending on a BT provided mac address of the phone. Moreover, it should not try the local bluetoothd MAC database as a fallback IIRC in this case.