nisargjhaveri/WirelessAndroidAutoDongle #283: Dongle keeps rebooting

dr-debanjan, 2025-06-15:
I have a Hyundai grand i10 nios 2024 model. I've used raspberry pi zero 2 W and flashed the latest image using the pi image tool. 

When connected to computer the device works & keeps looking for Android auto. When connected to the car unit the device boots up, the bluetooth is shown. But device auto reboots when tried to connect from phone. 

Tried multiple cables but same problem occurs. 

What might be the solution for the issue?

manio, 2025-06-15:
> When connected to the car unit the device boots up, the bluetooth is shown. But device auto reboots when tried to connect from phone.

I assume it is not moving but rather testing when the car is stopped?

dr-debanjan, 2025-06-15:
> Raspberry Pi Zero 2 W has dedicated USB power port so maybe try to additionally power it from some external source instead of relying on car HU USB and see if that helps...

Thank. Will try. Might need to something like #211 for now