nisargjhaveri/WirelessAndroidAutoDongle #284: Latest Android Auto Broken link

Ka0sHuNT3R, 2025-06-22:
So I have it up and running,  stop working since last android auto,  it sees the dongle but only connects to the Bluetooth.  It tries to connect to the wifi but soon disconnects.  Only way I have found to get it to work is uninstall updates then it connects and then asks to update which I do and it works but when I disconnect from car and come back later I have to go through the same steps of uninstalling updates and starting again

manio, 2025-06-22:
Hi,
I thought you have some broken URL (404) from the title ;)
What you described seems like some profile/persistent state problems. Is it OK when you clear the app data instead of uninstalling updates? Can you try to remove the whole bluetooth/phone profile in the car?

manio, 2025-07-16:
OK, my response was to @sorinvasilescu (forgot to mention)

LucianoCMPE, 2025-07-17:
This new img worked for a day now back to not working. It connects to BT but does not do the wifi latching and no AA on dash.