nisargjhaveri/WirelessAndroidAutoDongle #287: Wireless Auto Connect after BT Pairing

jplamb13, 2025-06-28:
When i connect to BT my phone doesnt prompt me to connect to the wifi network and doesnt do it automatically. Am i doing something wrong?

would me having the AA Wireless app installed be the issue?

manio, 2025-06-28:
You only need to connect/pair to bluetooth `WirelessAADongle-xxx` for the first time, rest is done automatically (no need to connect to WiFi).

EmmaR53, 2025-07-18:
I needed to turn on 'Wireless' in phone Settings>Apps>AndroidAuto. I had never seen this setting previously. It seened to appear because it knew there was now a wirless connection available. Connections then worked automatically from then on.
When looking again in Settings>Apps>Android Auto for the wireless setting, I was unable to find it where I had previously seen it. I needed to go to Settings>Apps>AndroidAuto>ConfigureInAndroidAuto>Tap on 'Version' a number of times to enter 'Developer Mode'>Tap 3 vertical dots in top right corner>DeveloperSettings>Tick the WirelessAndroidAuto tick box.