nisargjhaveri/WirelessAndroidAutoDongle #290: Unable to flash file on SD card

abhay3474, 2025-06-29:
[image]

     I'm using pi zero 2w at first i tried 5.0 release so i got error image is not multiple of 512byte.so tried older build 4.1 it's shows same issue
I tried with etcher too it giver error at end of validating 

https://github.com/user-attachments/assets/e58f578f-2813-4a42-a17e-96aed4c7dae5

manio, 2025-06-29:
You are probably trying to flash compressed image. Those are xz-compressed:

[image]
You need to un-xz it first before flash (at least on Raspberry Pi imager according to the message).

abhay3474, 2025-06-29:
I just figured it out I'm new at this so i just renamed .xz to . Img i didn't extracted it now it's working fine. 
Thanks for the reply ☺️