nisargjhaveri/WirelessAndroidAutoDongle #297: Won't connect 2019 F-150 Sync

joscthomas, 2025-07-04:
For Ford Sync on my 2019 F-150, I get : Device does not offer any SYNC-compatible features.

When connecting, I get : 

Ford F150 can search for available Bluetooth Devices. To pair a Bluetooth Device with Ford F150 in this manner, you will need to make your Bluetooth Device discoverable. This is usually done via the Device's Bluetooth Settings, but individual Devices may differ.

I select : Continue

Then I get :

Begin the pairing process from your device and a PIN will appear.

OR

Discover Other Bluetooth Devices

I select : Discover Other Bluetooth Devices. 

The I get :

Device does not offer any SYNC-compatible features.

About SYNC
SYNC 3, Software Version: 3.4
Build 20136_PRODUCT

joscthomas, 2025-07-09:
> Joseph,
> You need to connect the raspberry via USB cable to your SYNC first, then
> connect your phone via bluetooth to WirelessAADongle-. The dongle is
> doing all the handshake for you. You don't need to do any PIN-based
> pairing, etc. If you want to know the whole process, I described it here
> if you like to read.

Thank you for getting back to me so quickly. I got it working. I don't
recall reading about connecting the Raspberry Pi to the phone. Here are the
steps I took that worked. Perhaps you want to add them to the read.me file:

1. Connect your Android phone to the vehicle's USB port using a USB cable
and confirm that Android Auto works with a wired connection.

2. Disconnect your phone from the vehicle by removing the USB cable.

3. Make sure your phone is paired to the head unit via Bluetooth.

4. Pair your phone to the Raspberry Pi. Look for a device labeled
'WirelessAADDongle' (mine was 'WirelessAADongle-2e69ab'). (Do not pair the
Raspberry Pi to the head unit of your vehicle.)

5. After 10 or 20 seconds or so, Android Auto should start on your head
unit. Android Auto will now start automatically when you start your
vehicle.

hornjt75862, 2025-07-09:
I was not seeing my Raspbery pi's WirelessAADDongle on either my phone or
vehicle, but am seeing the "HandsFreeLink"  that the Honda Pilot mentions
in its instructions.

hornjt75862, 2025-07-09:
I was wondering if there should be any log files that are kept on the
Raspberry Pi  that would allow me to investigate the issue. Looking at the
USB CARD after unplugging the Raspberry Pi I would expect there to be some
files that would assist in the debugging the issues.

manio, 2025-07-10:
By default the logs are temporary, you need to SSH-in to get them. Or you can use aa-proxy-rs if you like (I have the webserver option for downloading logs more easily if you don't like to SSH or you can also set it to save to SD card).
More info [here](https://github.com/manio/aa-proxy-rs/issues/28#issuecomment-3055694143).