nisargjhaveri/WirelessAndroidAutoDongle #298: Won't connect to my Volvo cars 2019 S60, 2020 XC90)

laurenslo, 2025-07-07:
Can't get zero 2w to connect all the way. Samsung S23 Ultra stays connected to AAWirelessDongle AP, but my Volvo's show Connect compatible device. I tried the different connection methods, but none are working. On my phone it sometimes says it connect to Android Auto, but not on my Volvo's. Other times it just says Looking for android auto. I've tried everything I can think of. Any help would be appreciated.

greywire, 2025-07-08:
Yeah I'm also not having much luck with my 2021 Toyota Highlander. Phone pairs fine, but plug into car and says USB device not found or something...

hornjt75862, 2025-07-09:
Honda Pilot 2022 & RP Zero W
I am not getting a Bluetooth connection. I just get the light coming on and nothing else. I wait for over a minute still nothing.

manio, 2025-07-10:
The main problem seems to be:
```aawgd[269]: USB Manager: Timeout waiting for accessory start request```
You sure you're using a good cable with data?
You may also try this: https://github.com/manio/aa-proxy-rs/issues/28#issuecomment-3055694143

hornjt75862, 2025-07-10:
> Honda Pilot 2022 & RP Zero W I am not getting a Bluetooth connection. I just get the light coming on and nothing else. I wait for over a minute still nothing.

Is it safe to say if I do not get the Bluetooth connection while powering RP Zero W using the power connector,  it is not going to work with the USB connector? Just trying to test me new setup.

manio, 2025-07-10:
> Is it safe to say if I do not get the Bluetooth connection while powering RP Zero W using the power connector, it is not going to work with the USB connector? Just trying to test me new setup.

Sorry but don't know what you mean... I am powering my Pi Zero2W using single OTG cable:
[image]

manio, 2025-07-11:
> I would love an SSH wi-fi connection to AAWirelessDongle, but that is also not showing up.
Thank you for all the assistants provided by this group and will continue to try and figure out the issue with any assistants this group can provide.

This is rather pointing to some fatal error. Raspberry device WiFi should be always visible (even without password set). Yesterday I had a report from a guy which had similar problem, turns out that this was a SD card issue. System was not booting properly...

manio, 2025-07-14:
@hornjt75862
@jakissajmon discovered same lack of wifi problem in case of his Rpi0W, you're also have this, right?