nisargjhaveri/WirelessAndroidAutoDongle #300: Android 16 July 5 patch woes

3xtant, 2025-07-10:
I don't think this is an issue with this project, but rather an Android 16 problem.

My dongle has been working great for months. Upgrading to a new Pixel a couple of months ago was flawless. My phone got Android 16 a couple of months ago, which was uneventful. However, since the July 5 security update, the wireless connection stopped working. There are [reports of Android Auto issues with 16](https://www.autoevolution.com/news/this-is-the-first-bug-plaguing-android-auto-users-on-android-16-253348.html), so for other Pixel owners, delay the July upgrade if you can until Google releases a fix (or fixes?).

I had a January release, which had been running without any issues as mentioned above. When the dongle stopped working, I upgraded to the latest release, but nothing changed. My phone is simply "looking for AA" for as long as the dongle is connected.

I cannot do anything until the August patch is released and I get to download it, which might take me to the end of the month. I will reply to this thread when that happens and update on the issue.

manio, 2025-07-11:
@toffyjan yes this is the same problem
I just opened your logs from https://github.com/manio/aa-proxy-rs/issues/28 :)
Oh you'd be helpful on discord to help solve this.... as I don't have any report from my users, but I'll ask about the Android 16...

3xtant, 2025-07-12:
Thank you both for your replies. So I guess my device probably got a corrupt file and then I installed a broken version, lol
I will try an old, working version and then try the latest with the proposed fix to see if that does it.

For completeness, I thought I would post my logs with the latest without any fixes. I sanitized the mac addresses
P.S. Funny that it tried to connect to my home wifi...
``` bash
Jan  1 00:00:03 buildroot syslog.info syslogd started: BusyBox v1.36.1
Jan  1 00:00:03 buildroot user.info console: Starting klogd: OK
Jan  1 00:00:03 buildroot kern.notice kernel: klogd started: BusyBox v1.36.1 (2025-03-09 08:48:52 UTC)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Linux version 6.1.61 (runner@fv-az1774-831) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot -g7a59c3a4) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 Sun Mar  9 09:20:07 UTC 2025
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] random: crng init done
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory policy: Data cache writeback
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x15800000, size 64 MiB
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Zone ranges:
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Movable zone start for each node
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Early memory node ranges
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 104545
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=00:00:00:00:00:00 vc_mem.mem_base=0x1ec0000
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory: 340196K/421888K available (7906K kernel code, 702K rwdata, 2184K rodata, 256K init, 498K bss, 16156K reserved, 65536K cma-reserved)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000007] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000056] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000158] bcm2835: system timer (irq = 27)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000943] Console: colour dummy device 80x30
Jan  1 00:00:03 buildroot kern.info kernel: [    0.001442] printk: console [tty1] enabled
Jan  1 00:00:03 buildroot kern.info kernel: [    0.001520] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060290] CPU: Testing write buffer coherency: ok
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060396] pid_max: default: 32768 minimum: 301
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060586] LSM: Security Framework initializing
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060795] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060866] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.062269] cgroup: Disabling memory control group subsystem
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064154] cblist_init_generic: Setting adjustable number of callback queues.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064237] cblist_init_generic: Setting shift to 0 and lim to 1.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064559] Setting up static identity map for 0x8220 - 0x8258
[... cut ...]
Jan  1 00:02:07 buildroot user.info aawgd[221]: Sent WifiStartRequest, messageId: 1, wrote 17 bytes
Jan  1 00:02:07 buildroot user.info aawgd[221]: Read WifiStartResponse. length: 2, messageId: 7
Jan  1 00:02:07 buildroot user.info aawgd[221]: Expected WifiInfoRequest, got 0-��^Q (7). Abort.
Jan  1 00:02:07 buildroot user.info aawgd[221]: Bluetooth launch sequence completed
Jan  1 00:02:08 buildroot kern.info kernel: [  128.578137] read descriptors
Jan  1 00:02:08 buildroot kern.info kernel: [  128.580366] read strings
```

manio, 2025-07-13:
@3xtant Please try this image: https://skyboo.net/temp/aa/aa-proxy-rs.zip
I have the feedback that is is working OK even after this Google change, so I'd be grateful if you could also confirm this.

Also clearing cache is helping - but only for first run after this - so this is of course not a solution...

closed 2025-07-13 by 3xtant

3xtant, 2025-07-15:
Glad it worked for you as well @rokfor2000 
FWIW, mine is still working today, no tweaking needed. I drove to multiple destinations, so several stop/start cycles.
I do have to say this version takes much longer to start than the January version I used to have, but it works well.
@manio is this image from your project or a new version of this one? I'm okay either way, just curious.
P.S. Okay **and** thankful, btw

rokfor2000, 2025-07-15:
So we're both ok the same Android Auto:
Android Auto: 14.6.652454-release (146652454)

Both on Android 16, but different builds:
Me BP31.250523.010
Her BP2A.250705.008

But then again, just because hers worked doesn't mean it wouldn't have stopped at some point. Mine was working fine for some time.

3xtant, 2025-07-17:
There's something that occurred to me yesterday that is different in Android 16. Google implemented its VPN offer as part of the OS.

This was on by default in my phone and I had selected multiple WiFi networks where it should not turn on. The dongle worked okay here at home, where I tested it, but not at work. I think the VPN tried to connect through the AA WiFi connection and was breaking everything.

Mine is now off by default and my dongle is quite stable, just a couple of hiccups.
[image]