nisargjhaveri/WirelessAndroidAutoDongle #302: Now not working :(

Joeboyc2, 2025-07-12:
Ive been using a raspberryz2w and this app for the last few months and things have been working fine and until yesterday :(

The Android app no longer loads on my headunit, and when I check my phone, it is not connecting to the wife point, and periodically disconnects from the Bluetooth.

I have had this issue in the past and a quick restart of the pi fixes the issues, but unfortunately, not this time.

To try to resolve this, I decided to flash the later version 0.5.0, but it seems I have the same issue, so to troubleshoot this further, I have edited the config and have managed to connect to the pi via ssh and have stream the logs.

These seem to be the key points in the logs:

`Jan  1 00:02:39 buildroot user.info aawgd[268]: Trying to connect bluetooth device at path: /org/bluez/hci0/dev_34_E3_FB_0E_E5_DC
Jan  1 00:02:39 buildroot user.info aawgd[268]: Bluetooth device already connected, disconnecting
Jan  1 00:02:43 buildroot kern.info kernel: [  163.874188] read descriptors
Jan  1 00:02:43 buildroot kern.info kernel: [  163.875635] read strings
Jan  1 00:03:00 buildroot user.info aawgd[268]: Found 1 bluetooth devices
Jan  1 00:03:00 buildroot user.info aawgd[268]: Trying to connect bluetooth device at path: /org/bluez/hci0/dev_34_E3_FB_0E_E5_DC
Jan  1 00:03:00 buildroot user.info aawgd[268]: Bluetooth device already connected, disconnecting
Jan  1 00:03:19 buildroot daemon.err bluetoothd[171]: Device is already marked as connected
Jan  1 00:03:20 buildroot user.info aawgd[268]: AA Wireless NewConnection
Jan  1 00:03:20 buildroot user.info aawgd[268]: Path: /org/bluez/hci0/dev_34_E3_FB_0E_E5_DC, fd: 15
Jan  1 00:03:20 buildroot user.info aawgd[268]: Sending WifiStartRequest (ip: 10.0.0.1, port: 5288)
Jan  1 00:03:20 buildroot user.info aawgd[268]: Sent WifiStartRequest, messageId: 1, wrote 17 bytes
Jan  1 00:03:20 buildroot user.info aawgd[268]: Read WifiStartResponse. length: 2, messageId: 7
Jan  1 00:03:20 buildroot user.info aawgd[268]: Expected WifiInfoRequest, got `��u^Q (7). Abort.
Jan  1 00:03:20 buildroot user.info aawgd[268]: Bluetooth launch sequence completed
Jan  1 00:03:23 buildroot kern.info kernel: [  203.917226] read descriptors
Jan  1 00:03:23 buildroot kern.info kernel: [  203.918544] read strings
Jan  1 00:03:25 buildroot user.info aawgd[268]: Found 1 bluetooth devices
Jan  1 00:03:25 buildroot user.info aawgd[268]: Trying to connect bluetooth device at path: /org/bluez/hci0/dev_34_E3_FB_0E_E5_DC
Jan  1 00:03:25 buildroot user.info aawgd[268]: Bluetooth device already connected, disconnecting
Jan  1 00:03:28 buildroot daemon.err bluetoothd[171]: src/profile.c:ext_io_disconnected() Unable to get io data for AA Wireless: getpeername: Transport endpoint is not connected (107)
Jan  1 00:03:30 buildroot daemon.err bluetoothd[171]: src/device.c:search_cb() 34:E3:FB:0E:E5:DC: error updating services: Operation already in progress (114)
Jan  1 00:03:31 buildroot user.info aawgd[268]: AA Wireless NewConnection
Jan  1 00:03:31 buildroot user.info aawgd[268]: Path: /org/bluez/hci0/dev_34_E3_FB_0E_E5_DC, fd: 16
Jan  1 00:03:31 buildroot user.info aawgd[268]: Sending WifiStartRequest (ip: 10.0.0.1, port: 5288)
Jan  1 00:03:31 buildroot user.info aawgd[268]: Sent WifiStartRequest, messageId: 1, wrote 17 bytes
Jan  1 00:03:31 buildroot user.info aawgd[268]: Read WifiStartResponse. length: 2, messageId: 7
Jan  1 00:03:31 buildroot user.info aawgd[268]: Expected WifiInfoRequest, got ؏�u^Q (7). Abort.
Jan  1 00:03:31 buildroot user.info aawgd[268]: Bluetooth launch sequence completed
Jan  1 00:03:34 buildroot daemon.info hostapd: wlan0: STA 6c:6a:77:55:95:65 IEEE 802.11: disassociated
Jan  1 00:03:34 buildroot daemon.info hostapd: wlan0: STA 6c:6a:77:55:95:65 IEEE 802.11: disassociated
Jan  1 00:03:34 buildroot daemon.info hostapd: wlan0: STA 6c:6a:77:55:95:65 IEEE 802.11: associated
Jan  1 00:03:34 buildroot daemon.info hostapd: wlan0: STA 6c:6a:77:55:95:65 WPA: pairwise key handshake completed (RSN)
Jan  1 00:03:34 buildroot daemon.info dnsmasq-dhcp[233]: DHCPREQUEST(wlan0) 10.0.0.20 6c:6a:77:55:95:65
Jan  1 00:03:34 buildroot daemon.info dnsmasq-dhcp[233]: DHCPACK(wlan0) 10.0.0.20 6c:6a:77:55:95:65 Joeboy-YogaSlim7
Jan  1 00:03:40 buildroot daemon.err bluetoothd[171]: src/profile.c:ext_io_disconnected() Unable to get io data for AA Wireless: getpeername: Transport endpoint is not connected (107)
Jan  1 00:03:40 buildroot user.info aawgd[268]: AA Wireless NewConnection
Jan  1 00:03:40 buildroot user.info aawgd[268]: Path: /org/bluez/hci0/dev_34_E3_FB_0E_E5_DC, fd: 17
Jan  1 00:03:40 buildroot user.info aawgd[268]: Sending WifiStartRequest (ip: 10.0.0.1, port: 5288)
Jan  1 00:03:40 buildroot user.info aawgd[268]: Sent WifiStartRequest, messageId: 1, wrote 17 bytes
Jan  1 00:03:40 buildroot user.info aawgd[268]: Read WifiStartResponse. length: 2, messageId: 7
Jan  1 00:03:40 buildroot user.info aawgd[268]: Expected WifiInfoRequest, got @��u^Q (7). Abort.
Jan  1 00:03:40 buildroot user.info aawgd[268]: Bluetooth launch sequence completed
Jan  1 00:03:49 buildroot daemon.err bluetoothd[171]: src/profile.c:ext_io_disconnected() Unable to get io data for AA Wireless: getpeername: Transport endpoint is not connected (107)
Jan  1 00:03:49 buildroot user.info aawgd[268]: AA Wireless NewConnection
Jan  1 00:03:49 buildroot user.info aawgd[268]: Path: /org/bluez/hci0/dev_34_E3_FB_0E_E5_DC, fd: 18
Jan  1 00:03:49 buildroot user.info aawgd[268]: Sending WifiStartRequest (ip: 10.0.0.1, port: 5288)
Jan  1 00:03:49 buildroot user.info aawgd[268]: Sent WifiStartRequest, messageId: 1, wrote 17 bytes
Jan  1 00:03:49 buildroot user.info aawgd[268]: Read WifiStartResponse. length: 2, messageId: 7
Jan  1 00:03:49 buildroot user.info aawgd[268]: Expected WifiInfoRequest, got ���u^Q (7). Abort.
Jan  1 00:03:49 buildroot user.info aawgd[268]: Bluetooth launch sequence completed
Jan  1 00:03:50 buildroot user.info aawgd[268]: Found 1 bluetooth devices
Jan  1 00:03:50 buildroot user.info aawgd[268]: Trying to connect bluetooth device at path: /org/bluez/hci0/dev_34_E3_FB_0E_E5_DC
Jan  1 00:03:50 buildroot user.info aawgd[268]: Bluetooth device already connected, disconnecting
Jan  1 00:03:53 buildroot daemon.err bluetoothd[171]: src/profile.c:ext_io_disconnected() Unable to get io data for AA Wireless: getpeername: Transport endpoint is not connected (107)
Jan  1 00:03:54 buildroot user.info aawgd[268]: AA Wireless NewConnection
Jan  1 00:03:54 buildroot user.info aawgd[268]: Path: /org/bluez/hci0/dev_34_E3_FB_0E_E5_DC, fd: 19
Jan  1 00:03:54 buildroot user.info aawgd[268]: Sending WifiStartRequest (ip: 10.0.0.1, port: 5288)
Jan  1 00:03:54 buildroot user.info aawgd[268]: Sent WifiStartRequest, messageId: 1, wrote 17 bytes
Jan  1 00:03:55 buildroot user.info aawgd[268]: Read WifiStartResponse. length: 2, messageId: 7
Jan  1 00:03:55 buildroot user.info aawgd[268]: Expected WifiInfoRequest, got ���u^Q (7). Abort.
Jan  1 00:03:55 buildroot user.info aawgd[268]: Bluetooth launch sequence completed
`

Specifically, the last few lines seem to point to the issue. is there something i need to do with the wifi on my phone to sort this out, or is it a config change i have made that causes this?

manio, 2025-07-12:
Can you please try this image?
https://github.com/manio/aa-proxy-rs/actions/runs/16234601041/artifacts/3517959707

Joeboyc2, 2025-07-12:
Hey @manio,
yeh i'll give it a try

Joeboyc2, 2025-07-12:
Hey @manio,
i downloaded your artifact, but it was a singe file, not an image, so i went to your repo and grabbed the images from :https://github.com/manio/aa-proxy-rs/releases/tag/v0.6.0
However, this seems objectively worse :\ As I am not really getting much in the way of output in the logs, I also tried with my wife phone and she connects to the Bluetooth, but AA doesn't load either.

Is there any further debugging I can do here, or a way I can run the connection on my laptop, as it s a hassle to keep going to my car and back again

manio, 2025-07-12:
> i downloaded your artifact, but it was a singe file, not an image

It is a zip with all images inside. OK, I downloaded this for you and here you are again:
https://skyboo.net/temp/aa/aa-proxy-rs.zip

The release v0.6.0 you mentioned doesn't have a fix for this specific problem, so it won't work for sure.

Then if you want to debug:
enable static wifi password, connect to raspberry pi wifi and open http://10.0.0.1 in your browser.

Joeboyc2, 2025-07-12:
Ah sorry, your right, I just unzipped the first download :| 
I'll try this new image to see if I get a different result

Joeboyc2, 2025-07-12:
Nice, this new image works for me, awesome cheers dude and knowing that url could be useful in the future, thank you

Joeboyc2, 2025-07-12:
Yeh I'll grab a fresh set when I get home, I a few hours.

Just click the download logs button in the ui? Or copy from ssh?

manio, 2025-07-12:
> Just click the download logs button in the ui? Or copy from ssh?

No matter, clicking is easier :)
Of course do it after you successfully connect AA first :)

itsjustasemicolon, 2025-07-13:
I too have been facing the issue of Pi Zero2W not connecting to HU sometimes (not always) and the only way to fix that has been to unplug the Pi from the car and replug it.
Should I try the https://skyboo.net/temp/aa/aa-proxy-rs.zip image too @manio ?

manio, 2025-07-13:
> I too have been facing the issue of Pi Zero2W not connecting to HU sometimes (not always) and the only way to fix that has been to unplug the Pi from the car and replug it. Should I try the https://skyboo.net/temp/aa/aa-proxy-rs.zip image too [@manio](https://github.com/manio) ?

Sure, go ahead!

itsjustasemicolon, 2025-07-13:
It seems to be connecting fine now... I'm curious, can you tell me what's different in this compared to v0.5.0?

jakissajmon, 2025-07-13:
@manio I had the same problem as @Joeboyc2 on my RPi zero W(V1.1) where the phone after connecting via bluetooth would just say "Looking for Android Auto..." and after some time disconnect. I tried the pi zero w image from the link you posted and now I can't even connect to it with bluetooth. What could be the problem and solution for this problem?

jakissajmon, 2025-07-13:
> [@jakissajmon](https://github.com/jakissajmon) what your logs are saying? Did you try to connect manually for the first time to the BT dongle after flashing? It is necessary...

@manio I'm currently unable to connect to it via wi-fi and am away from home so I can't connect it to a monitor, but I'll try to get the logs from the sd card. I am not able to connect to it via bluetooth, because it doesn't show up in my phone's bluetooth settings. I was able to do it before but i had the same issue as @Joeboyc2 and wasn't able to connect via wi-fi.

manio, 2025-07-13:
@jakissajmon
Jak Ci nie widzi nawet WiFi to coś ewidentnie jest nie tak... WiFi musi działać nawet jak mój program tam nie zastartował. Po starcie mojej apki na raspberrym powinien się podnieść profil Bluetooth i powinieneś również widzieć nazwę po bluetooth (pierwszy raz musisz do niego połączyć ręcznie). Próbój kilka razy - pierwsze starty są "najgorsze" bo wszystko musi się zestawić - czasem pomaga jakieś czyszczenie profili - np profilu urządzenia bluetooth, albo i nawet wyczyszczenie "profilu samochodu" już w samych ustawieniach AndroidAuto.
Odnośnie ostatniego posta: logi domyślnie zapisują się w RAM/pamięci i przepadają przy restarcie.
Generalnie spróboj wbić na WiFi (najpierw musisz se ustawić hasło bo inaczej sie nie połączysz) i potem w przeglądarce wpisz http://10.0.0.1 - masz tam cały webinterfejs - możesz se wtedy ustawić logowanie na partycję "stałą" - wszystko jest opisane w opcjach :)

jakissajmon, 2025-07-13:
@manio 
Podłączę w domu pod monitor, może czegoś więcej się dowiem. Odnośnie WiFi: AP nie wyświetla się w dostępnych sieciach. Wcześniej po prostu mogłem połączyć się z bluetooth, mimo to jednak AA nie działało. Skoro logi zapisują się tylko w pamięci RAM, nie mam obecnie możliwości uzyskania do nich dostępu. Z tego co wcześniej próbowałem z aawgd to nie działa zewnętrzna klawiatura USB, więc raczej nie będę mógł w ten sposób odczytać logów. Mogę spróbować też użyć skryptu, który skopiuje plik gdzieś na karcie, abym mógł go potem odczytać. Chodzi o plik `/var/log/aa-proxy-rs.log`?

jakissajmon, 2025-07-13:
[image]

[image]

jakissajmon, 2025-07-13:
@manio 
Starsza wersja całkowicie nie odpala, wywala kernel panic że nie może zamontować partycji, na monitorze pokazuje:

[image]

Spróbuję zmodyfikować `config.txt` tak, aby klawiatura działała i może uda mi się zdobyć logi, ale mogą one nie być pomocne z powodu błędów przez te modyfikacje.

jakissajmon, 2025-07-13:
Spróbowałem jeszcze raz, tym razem wypakowałem plik i wtedy wrzuciłem na kartę. Odpaliło, ale problem nadal występuje.

manio, 2025-07-14:
Tak podsumowując z tego co czytam... to ogolnie jakies dziwne rzeczy sie tam dzieją (szkoda że nie mam nawet w szufladzie tego PiZeroW, bo osobiście używam 0W2 a chętnie bym to przetestował i sprawdził)... np od początku sie zastanawiam czy błędy z `mmc` "Get command interrupt" są normalne, jeszcze bardziej sie zastanawiam gdy miałeś je też na starszym obrazie (no ale ręki sobie nie dam uciąć i może to akurat normalne a problemy mogły wynikać z błędów w samym obrazie - w sensie logicznym bo nikt tego nie sprawdził)...

`user.info console: Starting network: FAIL` to też już miałeś od pierwszych screenów i też mnie to zastanawia - może ten FAIL wynika z tego, że nie wstaje to WiFI i nie ma interfejsu (potwierdza to ten komunikat: `daemon.warn dnsmasq[186]: warning: interface wlan0 does not currently exist`)... ale dlaczego...?

btw zawsze rób `ifconfig -a` bo on Ci pokaże również nie postawione/nie skonfigurowane interfejsy, aczkolwiek w tym wypadku moze faktycznie nie być wlan0 w systemie....

Jeśli mimo wszystko chcesz z tym nadal walczyć to możemy popróbować o ile masz jeszcze siły... ale przydało by mi sie troche info którego mi brakuje/lub co warto spróbować:
1. do czego był ten raspberry? masz go od nowości? czy miałeś z nim jakieś problemy "sprzętowe" typu brak wifi w jakichś innych projektach?
2. czy to pierwsze próby odpalenia tego projektu w sensie AndroidAuto?
3. czy on działał wcześniej ale przestał na jakichś poprzednij wersjach, czy wogole startujemy od zera?
4. czy próbowałeś obrazów z tego repozytorium (https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/releases)? opieram się na nich w swoim projekcie ale i tu i u mnie cały czas coś się zmienia i może po prostu jakiś commit zepsuł support tutaj albo u mnie i nikt tego nie zauważył... jak możesz potestuj jakieś nawet starsze wersje - przede wszystkim czy wstaje wifi - to jest podstawa i tak jak pisałem jeśli image karty działa OK i poprawnie zastartował to WiFi powinno wstać automatycznie - powinieneś zobaczyć apeka w liście sieci - za to odpowiada proces `hostapd`...
5. masz dobry zasilacz? te błędy mogą wynikać z problemów z zasilaniem - sproboj innego albo wepnij Pi do jakiegoś innego portu USB - od biedy może to być nawet jakieś porządne USB z innego urządzenia albo pewnie ładowarka - chodzi tylko o to żeby potwierdzić że to nie problem zasilacza...
6. last but not least: karta SD... pytanie czy ona napewno jest OK... miałem mnóstwo zgłoszeń problemów tego typu co ty po czym na końcu magicznie się okazywało że kartę SD wystarczyło wymienić na nową bo tamta była uwalona i nie domagała - i ona może robić takie dziwne rzeczy... niestety raspberry jest dość kapryśny w tych sprawach.
7. Btw: w ostatnich zmianach było [wywalenie](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/commit/9b322bb830977f59e5d6466de959d3bae2848325) "customowego" firmware - nie wiem czy to mogłoby pomóc... ktoś tam widziałem że pisał że wifi działa dużo lepiej po tych zmianach - ale to napewno dotyczyło nowszego rpi - chyba 4-ki. Tak czy siak nie jest to stety/niestety jeszcze wrzucone w żaden "release"...

> Po usunięciu sterownika OTG udało mi się podpiąć klawiaturę i zdobyć logi:

No bez OTG nie ma to szans zadziałać dalej - ale zrobiles spoko bo przynajmniej w logach widzę że coś sie tam proboje dziać - jak widać moja binarka jest ok bo wstaje i daje poprawne komunikaty początkowe. Jak widać stanął na czekaniu na podłączenie komórki po bluetooth, co niestety nie nastąpi bo nie widzisz "radia" - a to jest w jednym chipie i skoro nie działa CI wifi to bluetooth też zapewne...

ps. jeśli masz może Discorda to może bedzie prościej i lepiej tam pogadać... i jak nie bedziemy tu śmiecić w tym wątku (i w dodatku po polsku?.... tak - wiem, wiem... ja zacząłem... bo jest szybciej jak nie muszę w głowie tłumaczyć... hehe ;)
(mój ID to `manio07` w razie czego...) a jak nie masz to też spoko... :)

jbahlman75, 2025-07-14:
> It is different project based on this, read more: https://github.com/manio/aa-proxy-rs?tab=readme-ov-file#about

Thanks this fixed my problem.  I am on Android 15 S22+ but using android auto beta.