nisargjhaveri/WirelessAndroidAutoDongle #307: Always have to connect to Bluetooth after a shutdown

M00DY-DEV, 2025-07-16:
Every time I power down the RPi, I have to manually look for the Bluetooth connection on my phone and then connect. Meaning, it doesn't show up as a previously connected device on my phone and thinks it's a new device every time it powers on.

Otherwise works great.

Tttyagi123, 2025-07-16:
Problem started after the latest android auto app update, i clear apps cache and it starts working again but have to find a fix for it

Tttyagi123, 2025-07-18:
This build din work for me either, it does allow the phone to attempt to connect to the head unit repeatedly but no prompt on the car display unfortunately

manio, 2025-07-18:
> This build din work for me either, it does allow the phone to attempt to connect to the head unit repeatedly but no prompt on the car display unfortunately

Hmmm... It would be great if you could join our Discord and with your logs I'll try to do my best to help you solve your connection problems (link on my main project page)...

M00DY-DEV, 2025-07-18:
Tried 0.7.0. The Bluetooth connection connects every time BUT only connected to the WiFi a couple of times. Most of the time the phone just keeps attempting to connect to WiFi.

manio, 2025-07-18:
> Tried 0.7.0. The Bluetooth connection connects every time BUT only connected to the WiFi a couple of times. Most of the time the phone just keeps attempting to connect to WiFi.

So is it working or not? I am lost.... You don't need to connect to WiFi until you don't want to configure/get logs. Normally you are connecting from phone to raspberrry Pi bluetooth.

> What's the discord server?

https://en.wikipedia.org/wiki/Discord
Invite link: https://discord.gg/c7JKdwHyZu

Tttyagi123, 2025-07-18:
I turned on usb debugging under developer option and since then it connects without an issue. Will keep you guys posted how it goes

Tttyagi123, 2025-07-19:
For me,turning on usb debugging fixed the issue. As of now, it has connected every single time since i turned on this feature and hope it works out well for others too. Another observation is, if i don't use car's call button to activate voice command and instead use "hey google" or "ok google" key words, the voice command works as well.

MattSecaur, 2025-07-22:
I have this problem, too. It seems to work fine if I pair my phone with the RPI (a Zero W2, in my case) each time I get in the car, but it does not connect automatically.
I tried turning on USB Debugging on my phone and that didn't do anything (not sure why it would). 
Is the best option to switch to aa-proxy?

MattSecaur, 2025-08-01:
I switched to aa-proxy and it seems to be working fine now.
https://github.com/aa-proxy/aa-proxy-rs/releases/tag/v0.7.0