nisargjhaveri/WirelessAndroidAutoDongle #309: Failing in toyota raize 2024

snake404, 2025-07-17:
Hi there. I'm getting the following error in the logs, tested in both usb ports in the raspberry even with 2 cables connected but nothing is working, here the logs:

```
Jan  1 00:03:03 buildroot user.info aawgd[272]: Found 2 bluetooth devices
Jan  1 00:03:03 buildroot user.info aawgd[272]: Trying to connect bluetooth device at path: /org/bluez/hci0/dev_my_phone_bluettoth_mac_address
Jan  1 00:03:03 buildroot user.info aawgd[272]: Bluetooth device already connected, disconnecting
Jan  1 00:03:03 buildroot user.info aawgd[272]: HSP HS RequestDisconnection
Jan  1 00:03:03 buildroot user.info aawgd[272]: Path: /org/bluez/hci0/dev_my_phone_bluettoth_mac_address
Jan  1 00:03:05 buildroot daemon.err bluetoothd[173]: src/profile.c:ext_io_disconnected() Unable to get io data for AA Wireless: getpeername: Transport endpoint is not connected (107)
Jan  1 00:03:05 buildroot daemon.err bluetoothd[173]: src/profile.c:ext_connect() HSP HS failed connect to my_phone_bluettoth_mac_address: Software caused connection abort (103)
Jan  1 00:03:05 buildroot user.info aawgd[272]: Failed to connect device at path: /org/bluez/hci0/dev_my_phone_bluettoth_mac_address
Jan  1 00:03:05 buildroot user.info aawgd[272]: Trying to connect bluetooth device at path: /org/bluez/hci0/dev_the_car_bluetooth_mac_address
Jan  1 00:03:05 buildroot user.info aawgd[272]: Bluetooth device already connected, disconnecting
Jan  1 00:03:06 buildroot user.info aawgd[272]: Failed to connect device at path: /org/bluez/hci0/dev_the_car_bluetooth_mac_address
Jan  1 00:03:06 buildroot user.info aawgd[272]: Failed to connect to any known bluetooth device
Jan  1 00:03:25 buildroot kern.info kernel: [  205.497692] read descriptors
Jan  1 00:03:25 buildroot kern.info kernel: [  205.499045] read strings
```

Any solution for this would be appreciated.

Thanks in advance.

manio, 2025-07-17:
Hi
For me it seems like some problem with bluetooth connection from the dongle to your phone. What if you connect the opposite, i mean try to connect once to your bluetooth dongle from phone...
Was it working before or you are trying this for the first time ever?

EmmaR53, 2025-07-18:
Maybe phone is already connected to Head Unit via Bluetooth for playing media from your phone. Try turning off any media apps you might have playing on the phone. It worked for me.