nisargjhaveri/WirelessAndroidAutoDongle #310: Bluetooth Connection

TomC0654, 2025-07-18:
I have a ‘17 Honda Ridgeline, Standard head unit, Galaxy S24, Android 15, PiZero 2W. 
I have encountered the following situation:
In my phone the Bluetooth connection recognized WirelessAADongle-0bf375. I was able to get it to pair, but could not get the dongle connected to Bluetooth. It either would not connect at all or on repeated attempts it would sometimes connect for a few seconds, then drop the connection. I am not sure what to try next.
Any thoughts on addressing this issue would be appreciated. Thank you.

TomC0654, 2025-07-18:
> > In my phone the Bluetooth connection recognized WirelessAADongle-0bf375. I was able to get it to pair, but could not get the dongle connected to Bluetooth. It either would not connect at all or on repeated attempts it would sometimes connect for a few seconds, then drop the connection. I am not sure what to try next.
> > Any thoughts on addressing this issue would be appreciated. Thank you.
> 
> So you have AndroidAuto on your car HU for a short while? Can you share your logs?

Sorry, but I am very new to all this. Where would I find the logs?

TomC0654, 2025-07-19:
> I had similar problem when the phone (Samsung A52) was already connected to head unit (not to android auto) to play media from phone wirelessly. Turning the music player off on the phone resolved the Bluetooth connection issue. Subau Impreza Hatch 2017. I only put the Wirless device together a couple of hours ago, so not a lot of experience as yet.
> 
> Even after Bluetooth connected sucessfully, Android Auto still did not start. I needed to go to phone Settings>Apps>Android Auto and there was an option to turn on Wireless Android Auto. I had never seen this settings before. Once I turned it on, Android Auto started after about 20 seconds. I tested turning car off and back on multiple times and Android Auto started each time after about 20 seconds. When looking again in Settings>Apps>Android Auto for the wireless setting, I was unable to find it where I had previously seen it. I needed to go to Settings>Apps>AndroidAuto>ConfigureInAndroidAuto>Tap on 'Version' a number of times to enter 'Developer Mode'>Tap 3 vertical dots in top right corner>DeveloperSettings>Tick the WirelessAndroidAuto tick box. Cheers Emma

I was able to get into Developer Mode and found that the WirelessAndroidAuto box was checked. Still no luck getting the Pi to connect to Bluetooth for more than a few seconds before the connection drops. Thank you for trying to help.

manio, 2025-07-20:
Hmmm... and is it any better with [this](https://github.com/aa-proxy/aa-proxy-rs/releases)?

TomC0654, 2025-07-20:
Thank you so much for taking the time to try to address my issue.
Understanding that this is a hobby for you, I do not expect you to
devote time to my peculiar situation. I tried the new file, but still
without success.
I had seen the commercial dongles, but was scared away by all the
configuration problems people were having. I saw an article on XDA and that
is what sent me to your solution. The article made it seem that it was as
easy as could be.
I have seen two different sets of instructions for implementing the dongle.
On XDA, after preparing the dongle, you simply plug the Pi into the usb
data port, then connect your phone via Bluetooth and then launch Android
Auto.
The other instructions have you plug your phone into the usb data port,
engage Android Auto, unplug the phone and then attach the dongle.
Under both of these instructions I cannot get Android Auto to launch. On
one occasion, after getting the dongle to connect, it stayed connected long
enough for me to get a message on my phone to activate Android Auto on the
head unit. Unfortunately, there is no option to do that. The only time I
get an option to activate Android Auto on the head unit is when my phone is
plugged into the data port.
Perhaps I have a bad Pi or an incompatible head unit. In any event, thank
you for your time. You should not have to fret over this.

Tom

manio, 2025-07-21:
@TomC0654
I don't feel fret in any way :)
Both instructions are fine. That is how it should be done. Can you please [download](https://github.com/aa-proxy/aa-proxy-rs?tab=readme-ov-file#embedded-web-interface) and give me the logs?

manio, 2025-07-22:
> Is it possible to create this for apple car play as well?

https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/199

TomC0654, 2025-07-23:
> Fist you enable wifi password. Here you have instruction: https://github.com/nisargjhaveri/WirelessAndroidAutoDongle?tab=readme-ov-file#getting-logs
> 
> Then You try to connect AA normally several times to collect logs, then you disconnect WiFi and Bluetooth, connect only WiFi (you are using your password which you set earlier), head to http://10.0.0.1 via your browser and click "Download logs".

I received your instructions and went to my vehicle to connect so I could get up to date logs. As you stated, I tried to connect AA normally several time to collect logs. I was surprised when on the third try it connected! I did not do anything different than before, but it works. Hopefully, it will continue to work. Thank you again.