nisargjhaveri/WirelessAndroidAutoDongle #317: Opel ASTRA K - not working

prophet2812, 2025-07-24:
I am using a Rasberry Pi zero w2.
As soon as I plug the Rasberry into the USB port and connect to the Rasberry via Bluetooth with my mobile phone, after a while the message “Connected to Android Auto” appears, but the car display does not switch over. When I select it manually on the car display, the message “I have to connect it via USB” appears. 

I have already checked the following:
-) Correct USB port (OTG)
-) Different cable
-) With and without the engine running
-) Deleted Bluetooth pairing on my phone and reconnected

I pulled the log via SSH, maybe you can help me.
[Log.txt](https://github.com/user-attachments/files/21416292/Log.txt)

Thanks!

prophet2812, 2025-07-24:
OK, it looks like the log is rewritten every time it starts, which is why you can't see it. I'll create a new log.

manio, 2025-07-24:
In my [project](https://github.com/manio/aa-proxy-rs/releases) you can easily set the log to persist via webinterface ;)

prophet2812, 2025-07-25:
@manio I have now used your version.
The web interface did not work, but I downloaded the new logs via SCP.

Everything looks fine on the mobile phone, and I can see from the logs that the mobile phone is connected. However, it still does not work.

[image]

[aa-proxy-rs.log](https://github.com/user-attachments/files/21434787/aa-proxy-rs.log)

[messages.log](https://github.com/user-attachments/files/21434799/messages.log)

manio, 2025-07-25:
Hmmm... your logs are very similar to @Prathamesh0606 and https://github.com/aa-proxy/aa-proxy-rs/issues/31.

Your bluetooth handshake (phone-raspberry connection) is perfectly fine but the problem starts when the raspberry is trying to communicate with the headunit over USB. Can you make sure that you're using a good cable with data pins?

Is it working when you connect your phone directly via USB cable to headunit?

Prathamesh0606, 2025-07-25:
On Linux, you could try to connect the Pi to the machine and check if `lsusb` command shows any device connected or if `sudo dmesg` command shows any USB connect/disconnect related logs

prophet2812, 2025-07-25:
Thanks, guys, it was indeed the cable.
I then tried 12 more (some of them completely new cables) and it worked fine with two of them.

@manio Do you know why the web interface isn't working? It wants to start with the IP 0.0.0.0, which of course can't work.

Last question: what's the best way to manage this with two cell phones?