nisargjhaveri/WirelessAndroidAutoDongle #319: Bluetooth unstable: TATA Safari: AA never connects with car

ashishkg-ux, 2025-07-27:
I have Headend unit in TATA SAFARI. using Android Auto on nothing phone 3a.
Bluetooth gets connected and so do wifi. 
however connection is unstable and have seen that 

a) sometimes, it says AA connected on phone but never reflects on HeadUnit.
b) Bluetooth connection between mobile and RASPI W Zero is very unstable.

Below are the logs
 
[aawirelessdonglelogs.txt](https://github.com/user-attachments/files/21452800/aawirelessdonglelogs.txt)

ashishkg-ux, 2025-07-27:
Also, please advise how to increase log level of all the modules so that i can provide detailed logs.
I tried on hostapd.conf to set log level to 0 but it gives error that it is read only.

sudo does not work either.

manio, 2025-07-27:
Seems like the logs are cut in the middle. You phone connected, then we don't know what happened next...

manio, 2025-07-27:
> Bluetooth connection between mobile and RASPI W Zero is very unstable.

so it sometimes works or you don't have it working ever?

ashishkg-ux, 2025-07-27:
It never worked with head end unit. 
Sometimes, phone connects with bluetooth to raspi, wifi connection also happens and get message on phone that AA is available.|

But head unit  never recognises it. 

Attached logs again. This time,  i tried to force Bluetooth connection twice from mobile. It connects for a while.
Does not show AA available any of these times... on phone
Does not  show Android connected on Head unit on these times.

[aawirelesslogs_set2.txt](https://github.com/user-attachments/files/21453607/aawirelesslogs_set2.txt)

ashishkg-ux, 2025-07-27:
Jan  1 00:02:13 buildroot user.info aawgd[221]: Tcp server accepted connection
Jan  1 00:02:13 buildroot kern.info kernel: [  133.552430] dwc2 20980000.usb: bound driver configfs-gadget.default
Jan  1 00:02:13 buildroot daemon.err bluetoothd[138]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager object upon power down
Jan  1 00:02:13 buildroot user.info aawgd[221]: USB Manager: Enabled default gadget
Jan  1 00:02:13 buildroot daemon.err bluetoothd[138]: src/profile.c:ext_io_disconnected() Unable to get io data for AA Wireless: getpeername: Transport endpoint is not connected (107)
Jan  1 00:02:13 buildroot daemon.err bluetoothd[138]: src/profile.c:ext_io_disconnected() Unable to get io data for HSP HS: getpeername: Transport endpoint is not connected (107)
Jan  1 00:02:13 buildroot user.info aawgd[221]: Bluetooth adapter was powered off
Jan  1 00:02:43 buildroot user.info aawgd[221]: USB Manager: Timeout waiting for accessory start request
Jan  1 00:02:43 buildroot user.info aawgd[221]: USB Manager: Disabled all USB gadgets
Jan  1 00:02:43 buildroot kern.info kernel: [  163.568909] android_work: did not send uevent (0 0 00000000)

manio, 2025-07-27:
Hmmm.... for me this seems like some USB problem with HU communication. Are you sure you are using good quality USB cable with data (not only power for charging)?

I had similar reports in my [aa-proxy-rs](https://github.com/aa-proxy/aa-proxy-rs) few times in recent days ;)