nisargjhaveri/WirelessAndroidAutoDongle #320: Raspberry pi 5

FalconXX1, 2025-07-27:
Hello would you please make an Image for the pi5

manio, 2025-07-27:
> ill give it a shot and reply Thanks thou

If you want to read more about it: https://github.com/aa-proxy/aa-proxy-rs?tab=readme-ov-file#-aa-proxy-rs

manio, 2025-07-28:
Sorry @FalconXX1 but I am not familiar with this "Imager". I am linux user and using simple `dd` is enough for me...
The image is 100% OK because I have one user for whom it's working fine. Ask ChatGPT how to save an image from your system - maybe Balena Etcher?

Tttyagi123, 2025-07-28:
> Thanks but im having some problems im using pi imager but its not working[image]
> 
> Its just stuck here

Use rufus to flash it, not imager

FalconXX1, 2025-07-28:
Sooo after using rufus im getting this [image]

Or the screen is just black

Tttyagi123, 2025-07-28:
> Sooo after using rufus im getting this [image]
> 
> Or the screen is just black

Rufus always works for me out of the box. You may try using balena as well if that suits ur config better.