nisargjhaveri/WirelessAndroidAutoDongle #321: Toyota 2020 Corolla connection issue

cslds, 2025-07-28:
I have a 2020 Toyota Corolla and when I plug in the Pi 2 W with the image flashed on it I get "Device not recognized" on my headend screen.
I have tried different cables but no change. I also tried it on my2022 Toyota Rav4 XLS and get the same message.
This seems like a Toyota issue but do you know anyway around this?
Is there a configuration I could change on the Pi to fix it?

hartmms, 2025-07-29:
I get similar message on my 2018 Ram 2500.   I can see the bluetooth name, but when I try to connect to it, it says "can't communicate".  I think the Pi2W keeps rebooting as LED will turn off, then on like during a boot.  Also, when I dismiss the "device not supported" on the Ram screen, it will pop back up after several seconds.

manio, 2025-07-29:
This version: https://github.com/aa-proxy/aa-proxy-rs/releases/tag/v0.7.0
has the option to disable legacy mode. You can try to enable this ... maybe it would work better. My car supports this mode - but this option is HU dependent.

greywire, 2025-08-04:
I have a 2020 highlander so it probably has the same or similar unit in it. I haven't been able to get it to work.

manio, 2025-08-04:
If you guys could provide me the logs then I could take a look what could be wrong...

greywire, 2025-08-04:
I spoke too soon. I tried that build you posted, and et voila, it works perfectly.

Thank you!