nisargjhaveri/WirelessAndroidAutoDongle #328: Connection Issues with Hyundai Grand i10 Nios 2022 Model

rishabhrpg, 2025-08-02:
# Connection Issues with Hyundai Grand i10 Nios 2022 Model
First of all, I really appreciate the efforts and contribution to open source. I'm very interested to use this for my car as this provides us the ability to extend use cases of Pi, which is fantastic!

## Device Information
- **Car**: Hyundai Grand i10 Nios 2022 model
- **Phone**: Iqoo 13 (Android)
- **Hardware**: Raspberry Pi Zero 2W

## Issue Description

I'm having connection issues with my Hyundai Grand i10 Nios 2022. The device works perfectly with Google's Display Head Unit Emulator but fails to maintain connection in the actual car.

## What I've tested so far

**Tested with Display Head Unit Emulator by Google:**
https://developer.android.com/training/cars/testing/dhu

- Works flawlessly
- Connects instantly to the emulator running on MacBook  
- No lags observed

**In the actual car:**
1. First I connect my Android phone (Iqoo 13) using a Type C cable - it connects instantly as always
2. Then used the same raspberry pi zero 2w and the same tested cable with the emulator
3. I can see the bluetooth name (AAWireless***) and when trying to connect, it seems like the pi is getting rebooted
4. I can't connect to SSH to get logs as it just reboots and disconnects the WiFi before I have a chance to read the logs

**With Connection Strategy: 2:**
- This time it was not getting rebooted and I was able to read the logs
- But not getting connected as well
- Trying it for a couple of times, I got connected once somehow but only for a second
- I saw AA in the display and also heard music for a second before it disconnected and did not connect again

## Log Output Attempt 1

```
➜  ~ ssh root@10.0.0.1
root@10.0.0.1's password:
# cat /var/log/messages
Jan  1 00:00:02 buildroot syslog.info syslogd started: BusyBox v1.36.1
Jan  1 00:00:02 buildroot user.info console: Starting syslogd: OK
Jan  1 00:00:02 buildroot user.info console: Starting klogd: OK
Jan  1 00:00:02 buildroot kern.notice kernel: klogd started: BusyBox v1.36.1 (2025-03-09 08:49:34 UTC)
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jan  1 00:00:02 buildroot kern.notice kernel: [    0.000000] Linux version 6.1.61-v7 (runner@fv-az1922-927) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot -g7a59c3a4) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 SMP Sun Mar  9 09:21:32 UTC 2025
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi Zero 2 W Rev 1.0
Jan  1 00:00:02 buildroot kern.notice kernel: [    0.000000] random: crng init done
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x15800000, size 64 MiB
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Zone ranges:
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000]   DMA      [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000]   Normal   empty
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Movable zone start for each node
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Early memory node ranges
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] percpu: Embedded 12 pages/cpu s18004 r8192 d22956 u49152
Jan  1 00:00:02 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: s18004 r8192 d22956 u49152 alloc=12*4096
Jan  1 00:00:02 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 104545
Jan  1 00:00:02 buildroot kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:A7:4E:8F vc_mem.mem_base=0x1ec0000
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Memory: 337224K/421888K available (9216K kernel code, 718K rwdata, 2316K rodata, 1024K init, 520K bss, 19128K reserved, 65536K cma-reserved)
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[... 395 lines cut ...]
Jan  1 00:00:10 buildroot user.info aawgd[270]: Path: /org/bluez/hci0/dev_E4_57_68_CD_F2_FA, fd: 12
Jan  1 00:00:10 buildroot user.info aawgd[270]: Sending WifiStartRequest (ip: 10.0.0.1, port: 5288)
Jan  1 00:00:10 buildroot user.info aawgd[270]: Sent WifiStartRequest, messageId: 1, wrote 17 bytes
Jan  1 00:00:10 buildroot user.info aawgd[270]: Read WifiInfoRequest. length: 0, messageId: 2
Jan  1 00:00:10 buildroot user.info aawgd[270]: Sending WifiInfoResponse (ssid: AAWirelessDongle, bssid: 88:a2:9e:1b:93:2d)
Jan  1 00:00:10 buildroot user.info aawgd[270]: Sent WifiInfoResponse, messageId: 3, wrote 55 bytes
Jan  1 00:00:10 buildroot user.info aawgd[270]: Read WifiStartResponse. length: 2, messageId: 7
Jan  1 00:00:10 buildroot daemon.err bluetoothd[171]: src/profile.c:record_cb() Unable to get HSP HS SDP record: Host is down
Jan  1 00:00:11 buildroot authpriv.info dropbear[279]: Child connection from 10.0.0.20:55468
Jan  1 00:00:11 buildroot authpriv.info dropbear[279]: Generated hostkey /etc/dropbear/dropbear_ed25519_host_key, fingerprint is SHA256:9P6tRiunXPCBO5bgB8KLXPT/zGgmWlW8aYEFHt00MVE
Jan  1 00:00:18 buildroot authpriv.notice dropbear[279]: Password auth succeeded for 'root' from 10.0.0.20:55468
Jan  1 00:00:20 buildroot daemon.info hostapd: wlan0: STA 8a:e2:cc:ca:b9:43 IEEE 802.11: associated
Jan  1 00:00:20 buildroot daemon.info hostapd: wlan0: STA 8a:e2:cc:ca:b9:43 WPA: pairwise key handshake completed (RSN)
Jan  1 00:00:22 buildroot daemon.info hostapd: wlan0: STA 8a:e2:cc:ca:b9:43 IEEE 802.11: disassociated
Jan  1 00:00:22 buildroot daemon.info hostapd: wlan0: STA 8a:e2:cc:ca:b9:43 IEEE 802.11: disassociated
Jan  1 00:00:22 buildroot daemon.info hostapd: wlan0: STA 8a:e2:cc:ca:b9:43 IEEE 802.11: disassociated
Jan  1 00:00:22 buildroot daemon.info hostapd: wlan0: STA 1e:9a:79:27:a0:4c IEEE 802.11: associated
Jan  1 00:00:23 buildroot daemon.info hostapd: wlan0: STA 1e:9a:79:27:a0:4c WPA: pairwise key handshake completed (RSN)
Jan  1 00:00:24 buildroot daemon.info dnsmasq-dhcp[233]: DHCPDISCOVER(wlan0) 8a:e2:cc:ca:b9:43
Jan  1 00:00:24 buildroot daemon.info dnsmasq-dhcp[233]: DHCPOFFER(wlan0) 10.0.0.2 8a:e2:cc:ca:b9:43
Jan  1 00:00:24 buildroot daemon.info dnsmasq-dhcp[233]: DHCPDISCOVER(wlan0) 8a:e2:cc:ca:b9:43
Jan  1 00:00:24 buildroot daemon.info dnsmasq-dhcp[233]: DHCPOFFER(wlan0) 10.0.0.2 8a:e2:cc:ca:b9:43
```

Any help or suggestions would be appreciated!

manio, 2025-08-02:
> I can see the bluetooth name (AAWireless***) and when trying to connect, it seems like the pi is getting rebooted

This seems like some power problems. You can try to use my [aa-proxy-rs](https://github.com/aa-proxy/aa-proxy-rs) with legacy mode disabled, this way it is less connect/reconnect sequences maybe it would help, but this is HU dependent so ymmv.

My project also has a wired mode + also desktop-head-unit support as well so you can easily check if it is working OK on your desktop head unit.

rishabhrpg, 2025-08-02:
Hi @manio I tried [aa-proxy-rs](https://github.com/aa-proxy/aa-proxy-rs) on the same hardware, I first connected it to macbook , used the web interface to disable the legacy mode and tried with the google's emulator and it worked. 
Then I tried with the vehicle, I could see my phone trying to connect to Android Auto but looks like the RPI was getting rebooted. I tried to login to web interface but could not get a connection. 

## Random things I tried.

- Connect the USB cable to PWR port and not USB port, this allowed RPI to stay alive and let me see the web ui but ofcourse it can not start AA. 
- Tried supporting RPI with PWR port with additional power from macbook USB ( just in case )  did not work, it did not start the web ui. 
- I just saw there is an option to change the logfile location to a persistant file, will try to get the logs this way. 

### I can see in the UI the option for wired connection. I just could not find details on how can I do so with RPI Zero 2W, 
apologies for my limited knowledege here. 

[image]

manio, 2025-08-02:
> apologies for my limited knowledege here.

not a problem :)

> Then I tried with the vehicle, I could see my phone trying to connect to Android Auto but looks like the RPI was getting rebooted. I tried to login to web interface but could not get a connection.

This may be because raspberry remembered your phone and it is trying to connect to it via bluetooth. This is interfering with 2.4GHz wifi and you could have problems connecting (make sure that you have `Radomize MAC` disabled in network setting in your phone). This is unfortunately a known hardware issue with Pi02W. Bluetooth and Wifi is not working reliable together :(

> I just saw there is an option to change the logfile location to a persistant file, will try to get the logs this way.

if it is rebooting then it doesn't help as the logs won't have time to save anyway :(

Powering from the second (POWER) usb port seems to be good way to test it...

> I can see in the UI the option for wired connection. I just could not find details on how can I do so with RPI Zero 2W

connect your phone to mac, obtain the VID (vendor ID) and paste it there to the config followed by `:0`