nisargjhaveri/WirelessAndroidAutoDongle #329: using multiple phones, for multiple drivers?

jfabernathy, 2025-08-03:
I had no issues installing this app on a RPI4 and using it in a Kia Telluride 2022. 

However, I would like to figure out a way so that when my wife drives the car she can easily connect to the Wireless AA.

Has anyone worked out a way to make this 2 drive-2 phone mode work?

manio, 2025-08-03:
Shouldn't be a much problem. Just connect the second phone and the Raspberry will round-robin all phones he has in his database and will connect to the first in range (but it may be some randomness if both are). If you like to fiddle more with this, I am planning to add a GPIO button for this kind of switching in my [aa-proxy-rs](https://github.com/aa-proxy/aa-proxy-rs) project.

jfabernathy, 2025-08-03:
I had some success with the second phone. I put the first phone in Airplane mode and setup the second per normal.  It had some difficulty the first time, but after power cycling the RPI4 and the car it worked fine.

I think it will be best that the phone not connected to the AA have it's WiFi switched off. That way the driver has AA as normal and the passenger still has Internet access via 5G Mobile data.

We'll test is a lot this week as we use the Waze traffic app a lot.

I'll follow what you are doing with your GPIO solution.