nisargjhaveri/WirelessAndroidAutoDongle #330: Intermittent Android Auto connectivity issues.

ashishkg-ux, 2025-08-03:
In my Tata safari, it is working most of the times with workaround.
once connected, it works fine till the card/Dongle starts again.

Setup:

Nothing phone 3a ----> wireless to AAandroidDongle ----->wired to Head Unit
Nothing phone 3a also connects to wired unit with Bluetooth.

Issue:

I see issue when 

I start the car, Nothing phone 3a connects to car first and then to AADongle.
in this case Android auto connection with Head Unit is not successful with error seen on Head Unit as "No response from Device"

However, it works when

I forget the Bluetooth connection on my phone towards Head unit.
In this case, it works fine by first connecting to AAdongle and then estabilishing Android auto connection to Head Unit.

Any other work around of fix I should try??

manio, 2025-08-03:
It smells like a known issue with recent AndroidAuto version. Can you try out one of these sdcard builds?
https://skyboo.net/temp/aa/aaproxy-br/
It is an [aa-proxy-rs](https://github.com/aa-proxy/aa-proxy-rs) based.

ashishkg-ux, 2025-08-04:
Thank you @manio. Will try and update the outcome

ashishkg-ux, 2025-08-05:
I tried. It fails to connect even the first time. "No response from Device" message seen on Head unit. 
Will try more again. Any tips on what the config should look like

ashishkg-ux, 2025-08-07:
I tried. It worked second time after i forgot all the Bluetooth profiles again. AA worked fine.

After this did not remove bluetooth profiles..

Later, tried again it did not work (after power cycle of device)
Observed that bluetooth got connected to AAproxy. But it did not connect to wifi. hence Android auto did not get connected.
Now manually connected to wifi and Android auto connected.. Bingo
<<Logs are not available for above scenario>>

**Next time** <<Logs are attached for scenario below>>

[20250807160016_aa-proxy-rs.log](https://github.com/user-attachments/files/21664028/20250807160016_aa-proxy-rs.log)

it did not connect. Bluetooth is connected to Aaproxy . 
Wireless did not auto connect. 
Manually connected wireless..
It did not work.

Now after some time (without power cycle of device)
Forget bluetooth of headend unit from phone.
After some time, it connected.

Hope logs are helpful for debugging

ashishkg-ux, 2025-08-09:
i have loaded the newest release what will be wifi password for aa-proxy-rs. Need to connect my laptop.
Default password of ConnectAAWirelessDongle does not work.

manio, 2025-08-09:
> i have loaded the newest release what will be wifi password for aa-proxy-rs. Need to connect my laptop. Default password of ConnectAAWirelessDongle does not work.

credentials are in Note here:
https://github.com/aa-proxy/aa-proxy-rs?tab=readme-ov-file#first-time-connection