nisargjhaveri/WirelessAndroidAutoDongle #335: 2017 Kia Soul Exclaim not displaying

supergirl022, 2025-08-19:
Hardware: 
1. 2017 Kia Soul! (exclaim) 
2. Google Pixel 7 Pro 
3. Pi Zero 2 W

Problem: Pixel connects to the pi with no problems however Kia head unit does not display the AA hud. Pixel shows Android Auto connected while simultaneously showing looking for Android Auto. 

What I have already tried:
1. Re-imaging the pi
2. Using two micro-usb cables (1 for power, 1 for data)
3. Updating Kia Nav software
4. Multiple different micro-usb cables(roughly 5 different cables including a new data cable) 
5. Various forum "fixes" that have not panned out. 

I will say I am brand new to this whole thing so I'm sure I'm missing a troubleshooting step but any assistance would be greatly appreciated. From what I can tell the power on the pi is fine, for about 30 seconds, maybe less, the light flashes green then turns solid until I either turn the car off or unplug the pi. 

[image]

supergirl022, 2025-08-26:
Thank you for responding. I tried this as you suggested unfortunately I had the same results.

wickedaverage, 2025-08-28:
I have the same issue on a 2017 Kia Sportage, exactly as you described.

harlock-99, 2025-08-31:
Hi,
I just setup today the Androidautowireless on a raspberry ,I got exactly same issue with my Kia Soul Ev ( usine raspberry pi zero 2w), after many tests, I found the USB cable supplied with the raspberry bought was not compliant with my car, I need test with 2 others  cable to got it works.
I am testing now on if any issue to use the raspberry longer, to see  if no déconnexion.

Hope it helps you to solve your issue.